jhu-dvrk/sawIntuitiveResearchKit #19: Build cisst/SAW/dVRK using catkin

adeguet1, 2015-01-12:
It would be nice to build cisst/SAW/dVRK using catkin for ROS along with all the dVRK/cisst ROS bridge classes.  Ultimately we could also push these packages to the OSRF build farms and get debian/Ubuntu binary packages build for us.   That would allow to install the dVRK code using `apt-get install`.

adnanmunawar, 2015-01-12:
This is an excellent idea, I had been meaning to propose this for a long time.

adeguet1, 2015-01-19:
This is getting almost usable.   See wiki doc: https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/CatkinBuild

It would be great if our ROS users could give this a try.

Important notes:
- this works with catkin build tools only, not with catkin_make
- I'd like to add dvrk-ros to the main cisst-saw project but we need to trim down that repository.  The CAD files that are not loaded at runtime should be moved to another repository.

adeguet1, 2015-03-25:
catkin/ROS build is now working with the catkin built tools.  This code is now under the development branch.

closed 2015-03-25 by adeguet1