jhu-dvrk/sawIntuitiveResearchKit #37: Update LSPB for non zero initial velocities

adeguet1, 2015-05-04:
The LSPB trajectory generator in cisstRobot always assumes the current velocity is 0.  This works fine when the user waits for a requested goal to be reached before sending a new one.  When the user send new goals continuously, the trajectory generator doesn't start from the current velocity.  If goals are sent very close to each other, the system keeps starting at a 0 velocity and never gets a chance to accelerate.

To avoid this, we need to change `robLSPB` to take into account a non 0 initial velocity.

pkazanzides, 2015-05-05:
This would be a nice feature.

But, it sounds like the LSPB trajectory is being used pre-emptively, which is not a good idea without this feature (i.e., every move should be allowed to decelerate to 0 before starting a new move).

The Reflexxes library is one possible alternative.

adeguet1, 2017-05-17:
Latest development branch now uses Reflexxes by default.  This should allow to start trajectories with non 0 initial velocities.

closed 2017-05-17 by adeguet1