jhu-dvrk/sawIntuitiveResearchKit #43: Standardized ROS interfaces

adeguet1, 2015-06-25:
The ROS bridges for the dVRK have grown organically and we've had some trouble standardizing the topics names as well as payloads.

# Topics

The latest master branch has the following topics.  Note that all publishers skip the "get" prefix while all subscribers have a "set" prefix.  Should we add the "get", drop the "set" or leave it as is?

## Periodic publishers
```
/position_joint_current (cisst_msgs::vctDoubleVec)
/position_joint_desired(cisst_msgs::vctDoubleVec)
/position_cartesian_current (geometry_msgs::Pose)
/position_cartesian_desired (geometry_msgs::Pose)
```

## Subscribers
```
/set_robot_state (std_msgs::String)
/set_position_joint (cisst_msgs::vctDoubleVec)
/set_position_goal_joint (cisst_msgs::vctDoubleVec)
/set_position_cartesian (geometry_msgs::Pose)
/set_position_goal_cartesian (geometry_msgs::Pose)
```

## Events
```
/error (std_msgs::String)
/warning (std_msgs::String)
/status (std_msgs::String)
/robot_state  (std_msgs::String)
/goal_reached (std_msgs::Bool)
````

## Foot pedal

In the last release, we pushed new names accidentally (with useless suffix state), I (Anton) propose the following topics (all are events with std_msgs::Bool):
```
/clutch
/coag (or mono, or mono_coag)
/bi (or bi_coag)
/camera
/camera_plus
/camera_minus
```
The topic names for the two right pedals is a bit confusing, classic daVinci have one pedal without label and the other one is "coag".  da Vinci S have "bi" and "mono".  I propose "bi_coag" and "mono_coag".

# Messages, i.e. payloads

We have tried different payloads, initially using ROS standard messages as much as possible.  I (Anton) had reservations using the ROS JointState message with uninitialized fields.  The JointState message contains 4 arrays, names, positions, velocities and efforts.  For a topic returning a position, we've decided to leave all but the position array empty.  I'm not a bit fan of this and I pushed a new message type cisst_msgs::vctDoubleVec.  The main reason is that there doesn't seem to be a general purpose Float64Array in ROS.  There is a multi-dimensional array but its implementation is totally unpractical for most users.

What I propose is to revert to JointState.  For this to work, I will have to modify the mts command for the desired joint position to use a prmPositionJointGet type instead of vctDoubleVec.  Otherwise, the templated conversion method will not be able to determine which field to fill in ROS JointState.

The main motivation to use ROS messages as much as possible is to make it simple to work on remote nodes without checking-out any cisst/saw code.  For example, the ROS matlab implementation already supports all standard ROS messages while adding the custom message (say cisst_msgs::vctDoubleVec) is not trivial.

# Namespace

All topics are prefixed with `/dvrk` by default, followed by the arm name and then the topic itself, e.g. `/dvrk/PSM1/error`.  For topics related to a given arm maybe we should create a sub-namespace, e.g. `/dvrk/PSM1/io/potentiometers_joint_current`.

# New topics

We need to add support for:
* Joint and cartesian velocities publishers on arm
* Add support JointState from a new prmStateJoint class, can publish current and desired.  No subscriber yet
* IO, ideally in namespace `/dvrk/<arm>/io/` - the goal is to provide low level access for data logging and debugging (e.g. pots, temperatures, digital IOs, ....), publishers only.
* Tele-operation controllers: enable/disable, errors, scale, orientation between master/slave
* Console

jimshenzhen, 2015-08-08:
I wonder if we can add time stamp to the pose data.
For example, instead of using geometry_msgs::Pose, we can use geometry_msgs::PoseStamped.
It is for synchronization of the robot data with sensor data from other sources.

adeguet1, 2015-09-11:
In development branch:
* prmStateJoint added and used to report current position/velocity/effort as well as desired position/effort from PID.
* prmStateJoint now mapped to the ROS JointState message so the ROS bridge doesn't use any custom cisst type anymore
* Teleoperation component now has bridge so one can monitor/control enable, scale, lock orientation/position

closed 2015-10-01 by adeguet1