jhu-dvrk/sawIntuitiveResearchKit #72: psm.close_jaw() function doesn't work in devel branch

zrwang, 2017-08-20:
Hi Anton,

The close_jaw() function doesn't work properly on my side. I updated the software environment to the latest versions on devel branch. I checked commit history, and it seems that you changed function `mtsIntuitiveResearchKitPSM::SetJawPosition`. Is that possible it's related to this?

Thanks!
Zerui

adeguet1, 2017-08-20:
It’s a known bug (at least for me) and I’m working on a fix before next release.   There’s a workaround for now, i.e. force the control mode to switch using a move command without any effect (i.e. move to current position).   I wrote some scripts to test/debug the dVRK stack mote systematically and the work around in show in Python:
https://github.com/jhu-dvrk/dvrk-ros/blob/devel/dvrk_python/scripts/dvrk_psm_test.py#L56

And Matlab:
https://github.com/jhu-dvrk/dvrk-ros/blob/devel/dvrk_matlab/test/test_psm_api.m#L13

I hope to fix this later this week.  We shouldn’t have to do a “dummy” move command for the jaw commands to work.

Anton

---
Anton Deguet
www.lcsr.jhu.edu - www.malonecenter.jhu.edu

zrwang, 2017-08-21:
Thank you for your information! Let me know if I can help.

Best regards,
Zerui

adeguet1, 2017-11-08:
Fixed in 1.5

closed 2017-11-08 by adeguet1