jhu-dvrk/sawIntuitiveResearchKit #89: Cannot Dynamically Read Robot State

TomShkurti, 2018-03-27:
It is not possible (that I know of) to read the robot_state value of a component- that topic only publishes when the state is changed. Thus, applications cannot read states through ROS and, for instance, only attempt to move the PSM when it has been homed. I would like to request that either:

1. The topic /dvrk/.../robot_state be published at a constant rate; ~2Hz should be sufficient for most things
2. A ros service be added that will return the current state when called
3. A ROS message be added that will cause the /dvrk/.../robot_state message to publish again when something is published to it (basically another implementation of 2).

adeguet1, 2018-03-28:
In the devel branch, this has been changed to use "latched" ROS topics so the latest message/status update is preserved for new ROS nodes.   See https://github.com/jhu-cisst/cisst-ros/issues/5

Can you test the development branch and close this ticket if it works for you?   I realize that testing the development branch is a bit scary and will likely require some config file updates.

adeguet1, 2018-03-28:
To use the development branch, you'll need to switch a whole collection of repositories to devel, see https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/Development.  Also, you might need to do a full "catkin clean --yes"

adeguet1, 2018-07-13:
I believe the current implementation with latching solves this issue and has been released in 1.6.

closed 2018-07-13 by adeguet1

TomShkurti, 2018-07-22:
Finally managed to track down our .cal files and complete the upgrade; latching works as desired.