jhu-dvrk/sawIntuitiveResearchKit #97: Supporting multiple PSMs per MTM

apekshaav, 2018-07-17:
We would like to have the following features:

- Adding multiple master/slave teleoperation pairs
- Option to switch between such pairs during runtime; preferably through a ROS topic 

These features were included on an earlier ticket (#35) but have not been implemented yet.

adeguet1, 2018-07-19:
Quick question:  Do we want to support more than 2 PSMs on a single MTM.  For example, drive all PSMs (1, 2, and 3) with a single MTM.

I think I will implement a single command to circle through the PSMs associated to a single MTM.   For example, the ROS application would have to publish to:  `/dVRK/console/circle_teleop_psm` with an `std_msg::String` for the MTM name (i.e. circle_teleop_psm("MTMR")).   Then the dVRK console node will publish a couple of messages with `std::msg::Bool` under /`dVRK/MTMR-PSM1/selected` (True) and `/dVRK/MTMR-PSM2` (False) so the end-user application can detect which pair is selected.

Does that make sense?  Any foreseeable issue or missing feature on your end as a user?

apekshaav, 2018-07-19:
I think that makes sense. Just one question: How would the code handle the same PSM being associated with both MTMs? In other words, should a PSM be paired to only one MTM at a time?

adeguet1, 2018-07-19:
I didn't really think about that case, is this something you need for your application?   I guess I'm influenced by the clinical system implementation where the operator can toggle which PSM is controlled on one of the MTMs using the foot pedals.    I'll check again on our clinical Si to see what the native controller supports.

adeguet1, 2018-08-01:
@apekshaav , please confirm that you have the features you need in the devel branch and if so, close this ticket.

apekshaav, 2018-08-01:
Yes, we have what we need in the devel branch. Closing this ticket, thank you Anton!

closed 2018-08-01 by apekshaav