jhu-dvrk/sawIntuitiveResearchKit #98: Desired Positions Initialized to Garbage Upon Enabling Direct Control

TomShkurti, 2018-07-22:
Every time PID direct control is enabled in dvrk_console_json, the desired position fields initialize to numbers other than the current positions of the joints. These numbers have no relation to the joint positions reported elsewhere and are often extreme values (represented as 360.00 or -360.00) or zero, leading me to believe that they are garbage memory and not anything coming from the PIDs or encoders (which, elsewhere in the program, agree to within expected precision). It is still possible to zero the desired angle fields or fill in the current positions manually to allow for graceful movement, but this was done automatically in previous versions.

adeguet1, 2018-07-23:
This is indeed not very practical.   I believe I just fixed the issue in sawControllers.   Make sure you switch all your branches to "devel" to test this.   See https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/Development#compatible-development-branches, more specifically the section:
```sh
cd ~/catkin_ws/src/cisst-saw
git submodule foreach git checkout devel
git submodule foreach git pull
git submodule foreach git submodule init
git submodule foreach git submodule update
cd ~/catkin_ws/src/dvrk-ros
git checkout devel
git pull
```

Also, I've changed the behavior of "maintain position" and "zero position" so these buttons wouldn't stop the PID controller.   So now, you can just "enable" and this will maintain current position.   If you apply an external force on the arm and press "maintain position" again, this will reset the goals to current position.   "zero position" is actually a bit dangerous since will apply full PID from wherever you are to 0.   I might actually remove that button.   Your thoughts @TomShkurti ?

TomShkurti, 2018-07-24:
Verified that the initialize-to-garbage issue is gone in devel.

As for the zero button a warning message like appears when direct control is engaged would almost certainly be a good idea, but I agree that it would probably be better to just remove the thing completely. Higher-level applications requiring/assuming that the PSM is in its zero configuration when they take control is something we've recognized from fairly early on as bad practice and so having a convenient way to satisfy that condition is really not useful.

adeguet1, 2018-07-24:
I checked-in a couple of changes:
* sawControllers: https://github.com/jhu-saw/sawControllers/commit/623415bdb7d7e6c4c59971bd6f769535664f8f4f
* sawRobotIO1394: https://github.com/jhu-saw/sawRobotIO1394/commit/110c31d4052d03d15e5b98a7e55d8b5d3cc8dca0

Can you check these and close this issue if everything works as desired?

TomShkurti, 2018-07-25:
Functionality is working as advertised.

closed 2018-07-25 by TomShkurti