jhu-dvrk/sawIntuitiveResearchKit #135: What is the 'zero position'?

RobertoRoos, 2021-07-07:
I feel a little silly asking this, but what is the zero or home position of the PSMs, ECM and MTMs?
I was going through the calibration manual where it's mentioned many times, but I couldn't find a reference to what those positions should be exactly.

I presume for the PSMs this with the 4 tool DOFs straight, the tool slide all the way up, the arm rolled straight and the pivot all the way back. Same for the ECMs. But for the MTMs I have no clue.

Thanks. And I'm sorry if this is not the right place to ask.

adeguet1, 2021-07-07:
Please use the dVRK slack channel for this type of question.

adeguet1, 2021-07-07:
One way to visually see where the zero position is, is to use the dVRK in simulation mode with rViz.  Try:

roslaunch dvrk_robot dvrk_arm_rviz.launch arm:=MTMR

(arm:= can be ECM, PSM1 as well)

In dVRK 2.0, you can go to the “arm” GUI tab, select “direct control” and then move the arm in joint space.

Anton

closed 2021-07-07 by adeguet1