jhu-dvrk/sawIntuitiveResearchKit #137: [Wiki] Mention `dvrk_arm_test.py` in examples

RobertoRoos, 2021-07-13:
Let's give this a try as sort of a wiki pull request.

I would suggest we mention the script `dvrk_arm_test.py` explicitly on the wiki, since this script was exactly what I was looking for when I was done calibrating. It also works very well as an initial hardware validation step.

See **https://github.com/RobertoRoos/sawIntuitiveResearchKit.wiki.git**, branch **feature/calibration_arm_test**.

```
diff --git a/.gitignore b/.gitignore
new file mode 100644
index 0000000..dbe9c82
--- /dev/null
+++ b/.gitignore
@@ -0,0 +1 @@
+.vscode/
\ No newline at end of file
diff --git a/Calibration.md b/Calibration.md
index 6daac00..fbf53ef 100644
--- a/Calibration.md
+++ b/Calibration.md
@@ -13,6 +13,7 @@
       * [4.3. Calibrating scales](#43-calibrating-scales)
       * [4.3. Calibrating offsets](#43-calibrating-offsets)
       * [4.4. Errors during the potentiometer calibration](#44-errors-during-the-potentiometer-calibration)
+   * [5. Testing calibration](#5-testing-calibration)
 
 <!-- Added by: anton, at: 2021-01-28T15:58-05:00 -->
 
@@ -354,3 +355,8 @@ Similar to the scales, there is usually no point to save the results of the seco
 There is a potential egg and chicken issue.   While trying to calibrate the potentiometers, the safety checks using the potentiometers are still active.   So if the original calibration (from `.cal` files) is way off, there will likely be some safety checks triggered while the robot arms are moving.
 
 To avoid this, you can disable the pot/encoder safety checks but you MUST first visually check that the encoders and potentiometers work fine (see https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/Debugging-Potentiometer-Issues#visual-checks), you can disable the pots/encoder safety checks manually during the calibration process.   To do so, ÔÇ£HomeÔÇØ the arm and then switch to the IO Qt Widget, check the ÔÇ£Direct controlÔÇØ box and approve.  Then uncheck ÔÇ£Use pot/encoder checkÔÇØ box.   At that point, you can start the scale calibration using the Python script.
+
+
+# 5. Testing calibration
+
+When finished calibrating, run any of the [examples](/jhu-dvrk/sawIntuitiveResearchKit/wiki/Examples) to test your actuator ([dvrk_arm_test.py](/jhu-dvrk/sawIntuitiveResearchKit/wiki/Examples#2-arm-test) is useful in particular).
diff --git a/ECM.md b/ECM.md
index 0173e62..8167be3 100644
--- a/ECM.md
+++ b/ECM.md
@@ -85,7 +85,7 @@ This is a **VERY IMPORTANT PROCEDURE**.  At that point, we don't have a utility
 * `ReleasedCurrent`
 * `EngagedCurrent`, though this one is easy, it should be set to 0.
 
-For this procedure we will use the [sawRobotIO1394QtConsole](/jhu-dvrk/sawIntuitiveResearchKit/wiki/Examples#2-sawrobotio1394qtconsole) program along with the ECM XML configuration file for your arm.  You will need to manually edit the XML file and between changes, test using the `sawRobotIO1394QtConsole` program.
+For this procedure we will use the [sawRobotIO1394QtConsole](/jhu-dvrk/sawIntuitiveResearchKit/wiki/Examples#3-sawrobotio1394qtconsole) program along with the ECM XML configuration file for your arm.  You will need to manually edit the XML file and between changes, test using the `sawRobotIO1394QtConsole` program.
 
 ![sawRobotIO1394QtConsole with brakes](/jhu-dvrk/sawIntuitiveResearchKit/wiki/assets/gui/dvrk-gui-ecm-console.png)
 
diff --git a/Examples.md b/Examples.md
index 8121b2a..e8af51a 100644
--- a/Examples.md
+++ b/Examples.md
@@ -1,18 +1,19 @@
 <!--ts-->
    * [1. How to run the examples](#1-how-to-run-the-examples)
-   * [2. sawRobotIO1394QtConsole](#2-sawrobotio1394qtconsole)
-      * [2.1. Usage](#21-usage)
-      * [2.2. Widgets](#22-widgets)
-      * [2.3. Testing the digital inputs](#23-testing-the-digital-inputs)
-      * [2.4. Testing sensors](#24-testing-sensors)
-      * [2.5. Testing motor current](#25-testing-motor-current)
-   * [3. sawIntuitiveResearchKitQtPID](#3-sawintuitiveresearchkitqtpid)
+   * [2. ArmTest](#2-arm-test)
+   * [3. sawRobotIO1394QtConsole](#3-sawrobotio1394qtconsole)
       * [3.1. Usage](#31-usage)
       * [3.2. Widgets](#32-widgets)
-   * [4. sawIntuitiveResearchKitQtConsoleJSON (<strong>and</strong> ROS dvrk_robot dvrk_console_json)](#4-sawintuitiveresearchkitqtconsolejson-and-ros-dvrk_robot-dvrk_console_json)
+      * [3.3. Testing the digital inputs](#33-testing-the-digital-inputs)
+      * [3.4. Testing sensors](#34-testing-sensors)
[... 101 lines cut ...]
+# 5. sawIntuitiveResearchKitQtConsoleJSON (**and** ROS dvrk_robot dvrk_console_json)
 
-## 4.1. Usage
+## 5.1. Usage
 
 Use `sawIntuitiveResearchKitQtConsoleJSON` to test the whole system with a position based tele-operation controller.  The command line parameters can be found using `sawIntuitiveResearchKitQtConsoleJSON` without options:
 ```
@@ -173,7 +189,7 @@ Assuming that you have set your path correctly and you are in the directory that
 
 The file `console.json` contains a description of your system, i.e. how many arms and which configuration files to use for your arms as well as teleoperation components (see [file formats](/jhu-dvrk/sawIntuitiveResearchKit/wiki/Configuration-File-Formats)).  Since the `console-xyz.json` files depend on your system, you will have to create your own so they can reference the correct arm serial numbers.  Many examples can be found in the `share` directory.  `sawIntuitiveResearchKitQtConsoleJSON` will parse the JSON console configuration file to determine which components should be created, including Qt Widgets.  This allows to handle many different configurations without writing any C++ code nor re-compile.  Note that a "sister" application is available for ROS in `dvrk_robot`, `dvrk_console_json`.  The ROS topics will also be automatically populated based on the content of the JSON configuration file.
 
-## 4.2. Widgets
+## 5.2. Widgets
 
 The application introduces a few new widgets.   Please note that all the homing logic is implemented in the high level components so the IO and PID widgets should only be used for monitoring and debugging.   You do not need to use the IO widget to power the boards, reset the encoders or bias the current based on current feedback.   These steps are all triggered in sequence when you hit the `Home` button on the console widget.      
 
@@ -190,7 +206,7 @@ The arm widget can be used to monitor messages specific to an arm, current 3D po
 
 The widgets for PSM and ECM tele-operation will show the current state of the corresponding components.
 
-### 4.3. Instructions
+### 5.3. Instructions
 
 You will first need to power and home the whole system.  To do so, just check the `Home` box on the first tab.  We recommend to remove the tools before homing but this is not necessary.   If the system doesn't power on properly (check message in the first GUI tab), make sure your E-Stop button is released.
```

RobertoRoos, 2021-07-13:
I've also added a little calibration table in the same branch:

[image]

adeguet1, 2021-07-13:
Is your "fork" private?   The link to https://github.com/RobertoRoos/sawIntuitiveResearchKit.wiki.git leads to a 404 page on my end.

RobertoRoos, 2021-07-13:
No, it's public. But that's the git clone URL. The visiting URL is https://github.com/RobertoRoos/sawIntuitiveResearchKit/wiki.
However, since I've committed in two branches, you can't view those through the web.

(Yeah, this is all not really working out how I hoped it would)

adeguet1, 2021-07-13:
We might want to have a quick zoom to iron out the processes if we decide to use forks/branches.  On my end, I cloned your repo (`sawIntuitiveResearchKitWiki`), checked out each branch of your repo and use a local diff tool (meld) on my Linux PC to manually merge with my clone of `sawIntuitiveResearchKit.wiki`.   And since I'm not that bright I also added some not so minor changes before committing (so now it's a real mess, see comment below).   At that point, considering GitHub is not making it easy to use their own tools on the wiki repository, I'm inclined to let user edit the wiki directly and I can just use simple history link available on github (e.g. https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/Calibration/_history).

One decision I just made is to remove the numbering in titles.  This way links to sections/subsections are less likely to be broken if a new section is added in the middle of a page.

One a side note, there is a small script in the wiki repository itself to generate the TOC (another feature I wish GH wiki would have).   The script usage is something like `./gh-md-toc.sh --insert Calibrations.md`.  The md file needs a couple of html tags first: `` and ``

RobertoRoos, 2021-07-13:
Thanks.

I agree the convenience of numbering is not worth the extra inconvenience.

Small reflection on our procedure: you don't have the exact commits I made, so my PR is sort of broken. Of course I can now just remove by branches.

I think an official fork might be good. You should be able to make a Github action to update another repo (e.g. the .wiki.git) automatically. Let me make an example, this sounds entertaining.

I'll close this issue as my stuff is in there now.

closed 2021-07-13 by RobertoRoos

RobertoRoos, 2021-07-13:
Got a working demo here with a fork and automatic updates: https://github.com/RobertoRoos/WikiTest

Merging a pull requests updates the real wiki.

Problem is this completely overrides the regular wiki edits. The official fork does a forced push, so any 'local' changes would get removed again. I don't think that would be acceptable.

Still, an official fork with manual updates could work.