jhu-dvrk/sawIntuitiveResearchKit #140: configGUI.m fails to generate ECM xml file

linhongbin, 2021-07-28:
Hi,

I am using [configGUI.m](https://github.com/jhu-dvrk/sawIntuitiveResearchKit/blob/master/share/configGUI.m) to generate xml file of ECM following the instructions [here](https://github.com/jhu-dvrk/sawIntuitiveResearchKit/wiki/XMLConfig),

However I got error when generating ECM xml as following:

```matlab
Unrecognized function or variable 'setAttribut'.

Error in configGenerator (line 312)
        X_BrakeReleasedCurrent.setAttribut    % Actuator.setAttribute('Pos1', 'ENC');

Error in configGUI>generateButton_Callback (line 150)
isOK = configGenerator(...

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in configGUI (line 45)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)configGUI('generateButton_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.

```

I have successfully generated PSM xmls using [configGUI.m](https://github.com/jhu-dvrk/sawIntuitiveResearchKit/blob/master/share/configGUI.m), but not ECM. Do you know how to fix it?

adeguet1, 2021-07-28:
Sorry about that, I accidentally modified this file and it made it to 2.0.   This is fixed in the devel branch: https://github.com/jhu-dvrk/sawIntuitiveResearchKit/commit/9dde0a5d86181acd74d2ec1b030a344159e329b3

There are a few other changes in the devel branch so you might just want to copy paste that specific file in the master branch.   I plan to release 2.1 fairly soon.

linhongbin, 2021-07-30:
Thanks Anton, glad to hear that. I will try to use devel branch. cheers!

adeguet1, 2021-07-30:
The main change in the devel branch is related to the instruments database.  Instruments are now identified using the `name:model[version]` (`version` is optional).   For you, if you had something like `LARGE_NEEDLE_DRIVER_400006` in your arm configuration file, you will need to replace it with `LARGE_NEEDLE_DRIVER:400006` -- note the `_` replaced by a `:`.

closed 2021-07-30 by adeguet1