jhu-dvrk/sawIntuitiveResearchKit #152: How to config my own dvrk console

fanke1213812138, 2022-01-14:
Dear teams,

I want config my own console.json. For example, I wan a console that can provide the simulated PSMs and the ECM. Now I have a console.json with simulated PSMs, and I can teleoperate with the MTMs. I want to add the simulated ECM in this console.json, but it can not work. Can you tell me how to config a console with simulated PSMS and ECM?  Thank you very much!!

adeguet1, 2022-01-14:
Hello,

This should be do-able.  There are examples of console files with the full patient cart simulated: https://github.com/jhu-dvrk/sawIntuitiveResearchKit/blob/master/share/console/console-patient-side-simulated.json

Dear teams,

I want config my own console.json. For example, I wan a console that can provide the simulated PSMs and the ECM. Now I have a console.json with simulated PSMs, and I can teleoperate with the MTMs. I want to add the simulated ECM in this console.json, but it can not work.

Can you explain what you mean by “it can not work”?  Do you have an example of what you tried?  Maybe copy/paste the error messages or attach the cisstLog-xxxx.txt file.

Anton

Can you tell me how to config a console with simulated PSMS and ECM? Thank you very much!!

fanke1213812138, 2022-01-14:
Thank you very much for your reply!

With the console-MTMR-PSM1-MTML-PSM2-Teleop-Simulated_nojointlimit.txt, I can teleoperate the simulated fake PSMs with the real MTMs. The GUI is shown as this picture.
[image]
Then I add the the ECM in this .json file as consle_PSM_ECM_simulated.txt. There are no errors, but I still get the above GUI without MTM-ECM.

Thank you very much!

[console-MTMR-PSM1-MTML-PSM2-Teleop-Simulated_nojointlimit.txt](https://github.com/jhu-dvrk/sawIntuitiveResearchKit/files/7872651/console-MTMR-PSM1-MTML-PSM2-Teleop-Simulated_nojointlimit.txt)
[consle_PSM_ECM_simulated.txt](https://github.com/jhu-dvrk/sawIntuitiveResearchKit/files/7872655/consle_PSM_ECM_simulated.txt)

adeguet1, 2022-01-14:
I can’t help but notice you’re using 1.7 which is a bit old.  I hope the following works on 1.7 but I’m not 100% sure.

Teleoperation for the ECM requires 2 MTMs and the following should be added to your configuration file:
"ecm-teleop":
{
"mtm-left": "MTML",
"mtm-right": "MTMR",
"ecm": "ECM"
}

See for full example: https://github.com/jhu-dvrk/sawIntuitiveResearchKit/blob/master/share/jhu-daVinci/console-SUJ-ECM-MTMR-PSM1-MTML-PSM2-PSM3-Teleop.json#L115

Documentation for 2.1 (might also be compatible with 1.7): https://github.com/jhu-dvrk/sawIntuitiveResearchKit/blob/master/share/jhu-daVinci/console-SUJ-ECM-MTMR-PSM1-MTML-PSM2-PSM3-Teleop.json#L115

Anton

---
Anton Deguet
Associate Research Engineer
www.lcsr.jhu.edu<http://www.lcsr.jhu.edu> - www.malonecenter.jhu.edu<http://www.malonecenter.jhu.edu>

[Screenshot from 2022-01-14 19-55-36]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F42082278%2F149570882-4947a5b7-7373-4259-9704-d7d45fdad796.png&data=04%7C01%7Canton.deguet%40jhu.edu%7C0662f6b544a5499edb5f08d9d791e593%7C9fa4f438b1e6473b803f86f8aedf0dec%7C0%7C0%7C637777843888372206%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=ZZLSrUv9in%2BbMRnh8%2F0Ahvan%2BqO6ryRX7u0nu7oWGbw%3D&reserved=0>

adeguet1, 2022-01-14:
Oops, the last link was incorrect.  Documentation for JSON format: https://dvrk.lcsr.jhu.edu/documentation/schemas/v2.1/dvrk-console.html#ecm-teleop

Anton

---
Anton Deguet
Associate Research Engineer
www.lcsr.jhu.edu<http://www.lcsr.jhu.edu> - www.malonecenter.jhu.edu<http://www.malonecenter.jhu.edu>

fanke1213812138, 2022-01-17:
Thank you very much for your help. You help a lot! And my lab is going to update the version of dvrk. Thank you very much!

closed 2022-01-17 by fanke1213812138