jhu-dvrk/sawIntuitiveResearchKit #186: ROS2: XLM Config/Version must be at least 5

frank20a, 2023-10-06:
We already have dVRK running with ROS and I am trying to upgrade to ROS2. I used our previous XML configuration files from ROS but I am getting the following error when running `ros2 run dvrk_robot  dvrk_console_json -j console-ALL.json`
```bash
E- File: osaXML1394.cpp Line: 51 -  Configure: Config/Version must be at least 5, version found is 4
File: /home/meastro/utwente-daVinci/sawRobotIO1394-ECM-28927.xml
Make sure you generate your XML files with the latest config generator.
```
I used our .cal files and the configGUI.m file found on `ros2_ws/src/cisst-saw/sawIntuitiveResearchKit/share` to generate new xml files but I am getting the same error there. I read the configGenerator.m file and it is hardcoded to write Version 4 on the XML (line 419) and the provided example XML file all write `<Config Version="4">` inside.

adeguet1, 2023-10-06:
For the next release we are deprecating the Matlab based configuration generator.  We introduced a new Python based program that supports the older controller (Classic/QLA1) as well as the upcoming ones (Clasic/DQLA and Si/dRAC).   The script should be installed in your path after using catkin build (ROS 1) or colcon build (ROS 2): dvrk-config-generator.py   (https://github.com/jhu-dvrk/sawIntuitiveResearchKit/tree/devel/applications/config-generator)

The script will generate the sawRobotIO.xml config file as well as  simple arm config file (e.g. PSM1-12345.json) and a simple console file for a single arm (e.g. console-PSM1.json).  Note that the configuration files are working for both ROS 1 and ROS 2.   Sorry the wiki has not been updated

frank20a, 2023-10-14:
Thank you this worked!

closed 2023-10-14 by frank20a