jhu-dvrk/sawIntuitiveResearchKit #194: Calibration and example with Ethernet connection

MissingLight, 2023-12-20:
I am using the Si arm and just start setting everything up. I was able to connect the controller using the Ethernet and ping the robot. 
However, the following calibration using `sawIntuitiveResearchKitSPotentiometersCalibration` seems I must use the Firewire connection. Does this can be done through Ethernet?

pkazanzides, 2023-12-20:
You should use Firewire for everything.  Ethernet support is not stable enough in this release of firmware (for FPGA V3).  I plan to fix it in Firmware Rev 9.

MissingLight, 2024-01-12:
> You should use Firewire for everything. Ethernet support is not stable enough in this release of firmware (for FPGA V3). I plan to fix it in Firmware Rev 9.

Thanks for the reply. I changed everything to Firewire and the communication is now working. However, when I try to run `sawIntuitiveResearchKitSPotentiometersCalibration`, it shows the error of brake amplifier. I cannot find any document that mention about this.

```
Error: failed to turn on brake amplifiers:
 - status:             0            0            0
 - desired:            1            1            1
```
I am using the Firmware Version 8 for Si arm, with the latest software release 2.2.1. Thanks for the help.

closed 2024-01-17 by adeguet1