jhu-dvrk/sawIntuitiveResearchKit #206: patient_cart.launch file not found when running the examples in the documentation on ROS2 Humble with Ubuntu 22.04

tushar-c, 2024-11-24:
My System Configuration

**OS**: Ubuntu 22.04 on WSL2 for Windows 11

**ROS Version**: ROS2 Humble

As the title suggests, I'm encountering the following issue when trying to run the examples (https://dvrk.readthedocs.io/en/latest/pages/software/compilation/ros2.html#testing)

Following all the steps without facing any issues up to the following command:

`ros2 launch dvrk_model patient_cart.launch generation:=Classic`

When I run the above command, I get the following error:

`file 'patient_cart.launch' was not found in the share directory of package 'dvrk_model' which is at '/home/tushar/ros2_ws/install/dvrk_model/share/dvrk_model'`

adeguet1, 2024-11-25:
I don't have access to a Linux box now but can you try:
```
ros2 launch dvrk_model patient_cart.launch.py generation:=Classic
```
I think the issue comes from the missing `.py` at the end of the launch file name.

tushar-c, 2024-11-25:
@adeguet1 thank you for your response. Replacing `patient_cart.launch` with `patient_cart.launch.py` in the above command fixed the issue for me! Maybe we can update the documentation as well regarding this.

closed 2024-11-25 by tushar-c