jhu-dvrk/sawIntuitiveResearchKit #221: overheating and temperature spikes

sph001, 2025-03-26:
intermittently when using our DVRK, we will see long streams of log output such as:

`11:30:32 warning #2014: IO: ECM controller measured temperature is 60.5ºC, warning threshold is set to 60ºC`

this will hold steady for 20-30 minutes, constantly at 60 +- 1 degree. then it will randomly start to build until it hits it's limit and faults relatively quickly.

`11:31:45 warning #2030: ECM IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:31:45 warning #2031: PSM1 IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:32:40 warning #2032: IO: ECM controller measured temperature is 62ºC, warning threshold is set to 60ºC
11:33:45 warning #2033: ECM IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:33:45 warning #2034: PSM1 IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:34:40 warning #2035: IO: ECM controller measured temperature is 63ºC, warning threshold is set to 60ºC
11:35:45 warning #2036: ECM IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:35:45 warning #2037: PSM1 IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:36:40 warning #2038: IO: ECM controller measured temperature is 63.5ºC, warning threshold is set to 60ºC
11:37:45 warning #2039: ECM IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:37:45 warning #2040: PSM1 IO: average compute time (0.98 ms) exceeds expected period (0.67 ms)
11:38:34 error #1: IO: ECM controller measured temperature is 64.5ºC, error threshold is set to 65ºC
11:38:34 error #2: ECM-PID: received [IO: ECM controller measured temperature is 64.5ºC, error threshold is set to 65ºC]
11:38:34 error #3: ECM: received [ECM-PID: received [IO: ECM controller measured temperature is 64.5ºC, error threshold is set to 65ºC]]
11:38:34 status #49: ECM: desired state FAULT
11:38:34 status #50: ECM: current state FAULT
11:38:34 error #4: ECM: detected power loss (actuators)
11:38:34 status #51: ECM: desired state FAULT
11:38:34 status #52: ECM: current state FAULT
11:38:35 error #5: IO: PSM1 power is unexpectedly off
11:38:35 error #6: PSM1-PID: received [IO: PSM1 power is unexpectedly off]
11:38:35 error #7: PSM1: received [PSM1-PID: received [IO: PSM1 power is unexpectedly off]]
11:38:35 status #53: PSM1: desired state FAULT
11:38:35 status #54: PSM1: control UNDEFINED_SPACE/UNDEFINED_MODE
11:38:35 status #55: PSM1: current state FAULT
11:38:35 error #8: PSM1: detected power loss (actuators)
11:38:35 status #56: PSM1: desired state FAULT
11:38:35 status #57: PSM1: current state FAULT`

I was hoping you might help diagnose why this might be happening. 
it only happens to the ECM on this specific DVSS.

adeguet1, 2025-03-26:
Can you share where you are?  There different versions of controllers on dVRK sites and it would help to know where you are.
Also, do you happen to know if your ECM controller is equipped with active fans on the QLAs.  See https://dvrk.readthedocs.io/2.3.0/pages/setup/Classic/modifications/qla-heat-sink.html

sph001, 2025-03-28:
this is the UCL-CBH dvss.

just took it apart and it does not appear to have the heat sinks installed, so I'll go about sorting that out. thank you!